m4nh/skimap_ros #1: segmentation error in slamdunk_tracker

gliese581gg, 2017-05-13:
Hi! I'm trying to use the slamdunk_tracker.

However, I got the error :

[slamdunk_tracker-1] process has died [pid 7893, exit code -11, cmd /home/user/catkin_ws/devel/lib/skimap_ros/slamdunk_tracker __name:=slamdunk_tracker __log:=/home/user/.ros/log/947ab95c-37a6-11e7-8cec-2c4d545321e9/slamdunk_tracker-1.log].
log file: /home/user/.ros/log/947ab95c-37a6-11e7-8cec-2c4d545321e9/slamdunk_tracker-1*.log

I found out that segmentation fault happens at slamdunk_tracker.cpp, line 84:

 slamdunk::SlamDunkScene slam_dunk_scene(slam_dunk);

How can I solve this problem?

m4nh, 2017-05-13:
@gliese581gg which version of PCL are you linking? Can you perform a GDB execution to track the segmentation fault to obtain a verbose log?

zhengguoxian123, 2017-06-22:
I also have the same error
[slamdunk_tracker-1] process has died [pid 7893, exit code -11, cmd /home/user/catkin_ws/devel/lib/skimap_ros/slamdunk_tracker __name:=slamdunk_tracker __log:=/home/user/.ros/log/947ab95c-37a6-11e7-8cec-2c4d545321e9/slamdunk_tracker-1.log].
log file: /home/user/.ros/log/947ab95c-37a6-11e7-8cec-2c4d545321e9/slamdunk_tracker-1*.log

I found out that segmentation fault happens at slamdunk_tracker.cpp, line 84:

closed 2018-05-21 by m4nh