m4nh/skimap_ros #3: Compiling error SLAMDUNK Eigen + G2O

AndreaIncertiDelmonte, 2017-05-28:
Hi,
I'm trying to compile skimap + slamdunk on Ubuntu 14.04 and ROS indigo and catkin tools.
I installed g2o into the folder **<catkinws>/skimap_ros/thirdparty/**, compiled the lib from source and added to CMakeList with `SET( G2O_ROOT  ${PROJECT_SOURCE_DIR}/thirdparty/g2o).`
I also installed opencv2.4 non free packages.
Which version of Eigen are you using?

I can compile SKIMAP but I have this issue when I try to include SLAM_DUNK:
`In file included from /usr/local/include/eigen3/Eigen/Core:343:0,
                 from /usr/local/include/eigen3/Eigen/Geometry:11,
                 from /home/andrea/development/ros/catkin_tools_ws/src/skimap_ros/include/slamdunk/graph_utils.h:13,
                 from /home/andrea/development/ros/catkin_tools_ws/src/skimap_ros/include/slamdunk/graph_backend.h:13,
                 from /home/andrea/development/ros/catkin_tools_ws/src/skimap_ros/src/slamdunk/graph_backend.cpp:9:
/usr/local/include/eigen3/Eigen/src/Core/CoreEvaluators.h: In instantiation of ‘Eigen::internal::evaluator<Eigen::PartialReduxExpr<ArgType, MemberOp, Direction> >::evaluator(Eigen::internal::evaluator<Eigen::PartialReduxExpr<ArgType, MemberOp, Direction> >::XprType) [with ArgType = const Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs_op<double>, const Eigen::Matrix<double, 0, 0, 0, 0, 0> >; MemberOp = Eigen::internal::member_sum<double>; int Direction = 0; Eigen::internal::evaluator<Eigen::PartialReduxExpr<ArgType, MemberOp, Direction> >::XprType = Eigen::PartialReduxExpr<const Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs_op<double>, const Eigen::Matrix<double, 0, 0, 0, 0, 0> >, Eigen::internal::member_sum<double>, 0>]’:
/usr/local/include/eigen3/Eigen/src/Core/Redux.h:338:95:   required from ‘Eigen::internal::redux_evaluator<_XprType>::redux_evaluator(const XprType&) [with _XprType = Eigen::PartialReduxExpr<const Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs_op<double>, const Eigen::Matrix<double, 0, 0, 0, 0, 0> >, Eigen::internal::member_sum<double>, 0>; Eigen::internal::redux_evaluator<_XprType>::XprType = Eigen::PartialReduxExpr<const Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs_op<double>, const Eigen::Matrix<double, 0, 0, 0, 0, 0> >, Eigen::internal::member_sum<double>, 0>]’
/usr/local/include/eigen3/Eigen/src/Core/Redux.h:416:35:   required from ‘typename Eigen::internal::traits<T>::Scalar Eigen::DenseBase<Derived>::redux(const Func&) const [with BinaryOp = Eigen::internal::scalar_max_op<double, double>; Derived = Eigen::PartialReduxExpr<const Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs_op<double>, const Eigen::Matrix<double, 0, 0, 0, 0, 0> >, Eigen::internal::member_sum<double>, 0>; typename Eigen::internal::traits<T>::Scalar = double]’
/usr/local/include/eigen3/Eigen/src/Core/Redux.h:438:73:   required from ‘typename Eigen::internal::traits<T>::Scalar Eigen::DenseBase<Derived>::maxCoeff() const [with Derived = Eigen::PartialReduxExpr<const Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs_op<double>, const Eigen::Matrix<double, 0, 0, 0, 0, 0> >, Eigen::internal::member_sum<double>, 0>; typename Eigen::internal::traits<T>::Scalar = double]’
/usr/local/include/eigen3/Eigen/src/LU/PartialPivLU.h:522:13:   required from ‘void Eigen::PartialPivLU<MatrixType>::compute() [with _MatrixType = Eigen::Matrix<double, 0, 0, 0, 0, 0>]’
/usr/local/include/eigen3/Eigen/src/LU/PartialPivLU.h:131:15:   required from ‘Eigen::PartialPivLU<MatrixType>& Eigen::PartialPivLU<MatrixType>::compute(const Eigen::EigenBase<OtherDerived>&) [with InputType = Eigen::Matrix<double, 0, 0, 0, 0, 0>; _MatrixType = Eigen::Matrix<double, 0, 0, 0, 0, 0>]’
/usr/local/include/eigen3/Eigen/src/LU/PartialPivLU.h:323:27:   [ skipping 5 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]
/usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:787:42:   required from ‘void Eigen::internal::call_assignment(Dst&, const Src&, const Func&, typename Eigen::internal::enable_if<(! Eigen::internal::evaluator_assume_aliasing<Src>::value), void*>::type) [with Dst = Eigen::Matrix<double, 0, 0, 0, 0, 0>; Src = Eigen::Inverse<Eigen::Matrix<double, 0, 0, 0, 0, 0> >; Func = Eigen::internal::assign_op<double, double>; typename Eigen::internal::enable_if<(! Eigen::internal::evaluator_assume_aliasing<Src>::value), void*>::type = void*]’
/usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:765:93:   required from ‘void Eigen::internal::call_assignment(Dst&, const Src&) [with Dst = Eigen::Matrix<double, 0, 0, 0, 0, 0>; Src = Eigen::Inverse<Eigen::Matrix<double, 0, 0, 0, 0, 0> >]’
/usr/local/include/eigen3/Eigen/src/Core/PlainObjectBase.h:710:65:   required from ‘Derived& Eigen::PlainObjectBase<Derived>::_set(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::Inverse<Eigen::Matrix<double, 0, 0, 0, 0, 0> >; Derived = Eigen::Matrix<double, 0, 0, 0, 0, 0>]’
/usr/local/include/eigen3/Eigen/src/Core/Matrix.h:225:30:   required from ‘Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>& Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>::operator=(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::Inverse<Eigen::Matrix<double, 0, 0, 0, 0, 0> >; _Scalar = double; int _Rows = 0; int _Cols = 0; int _Options = 0; int _MaxRows = 0; int _MaxCols = 0]’
/home/andrea/development/ros/catkin_tools_ws/src/skimap_ros/thirdparty/g2o/g2o/core/block_solver.hpp:370:10:   required from ‘bool g2o::BlockSolver<Traits>::solve() [with Traits = g2o::BlockSolverTraits<6, 0>]’
/home/andrea/development/ros/catkin_tools_ws/src/skimap_ros/src/slamdunk/graph_backend.cpp:417:1:   required from here
/usr/local/include/eigen3/Eigen/src/Core/util/StaticAssert.h:32:40: error: static assertion failed: EIGEN_INTERNAL_ERROR_PLEASE_FILE_A_BUG_REPORT__INVALID_COST_VALUE
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
                                        ^
/usr/local/include/eigen3/Eigen/src/Core/util/StaticAssert.h:214:7: note: in expansion of macro ‘EIGEN_STATIC_ASSERT’
       EIGEN_STATIC_ASSERT((C)>=0 && (C)<=HugeCost*HugeCost, EIGEN_INTERNAL_ERROR_PLEASE_FILE_A_BUG_REPORT__INVALID_COST_VALUE);
       ^
/usr/local/include/eigen3/Eigen/src/Core/CoreEvaluators.h:1280:5: note: in expansion of macro ‘EIGEN_INTERNAL_CHECK_COST_VALUE’
     EIGEN_INTERNAL_CHECK_COST_VALUE(TraversalSize==Dynamic ? HugeCost : int(CostOpType::value));
     ^
/usr/local/include/eigen3/Eigen/src/Core/util/StaticAssert.h:32:40: error: static assertion failed: EIGEN_INTERNAL_ERROR_PLEASE_FILE_A_BUG_REPORT__INVALID_COST_VALUE
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
                                        ^
/usr/local/include/eigen3/Eigen/src/Core/util/StaticAssert.h:214:7: note: in expansion of macro ‘EIGEN_STATIC_ASSERT’
       EIGEN_STATIC_ASSERT((C)>=0 && (C)<=HugeCost*HugeCost, EIGEN_INTERNAL_ERROR_PLEASE_FILE_A_BUG_REPORT__INVALID_COST_VALUE);
       ^
/usr/local/include/eigen3/Eigen/src/Core/CoreEvaluators.h:1281:5: note: in expansion of macro ‘EIGEN_INTERNAL_CHECK_COST_VALUE’
     EIGEN_INTERNAL_CHECK_COST_VALUE(CoeffReadCost);
     ^
make[2]: *** [CMakeFiles/slamdunk.dir/src/slamdunk/graph_backend.cpp.o] Errore 1
make[1]: *** [CMakeFiles/slamdunk.dir/all] Errore 2
make: *** [all] Errore 2
`
Tnks
Andrea

m4nh, 2017-05-30:
@AndreaIncertiDelmonte we use the Eigen3 version. Seems that G2O was compiled with a different version of Eigen w.r.t. the one used to compile the Slamdunk library. If you have problem with slamdunk feel free to use Skimap without it for initial tests

closed 2017-06-08 by m4nh

adnion, 2017-06-25:
@m4nh: I have exactly the same issue. I am quite sure that G2O was compiled using same version of Eigen which I also use for building SkiMap. Please explain how to resolve this problem.

m4nh, 2017-06-26:
@adnion please try to best describe your hardware setup so i can reproduce or find the solution