m4nh/skimap_ros #6: Issues for 3D map generation using gazebo and turtulebot

AmirCognitive, 2017-06-08:
Hi,

Thanks for your package.

I am having some issues generating the correct 3D map as you can see in the following image. Could you explain what do you think is the problem?

[image]

contents of the lunch file:

<launch>

    <arg name="camera" default="camera" />

    <node name="skimap_live" output="screen" pkg="skimap_ros" type="skimap_live">
        
        
        <param name="hz" value="10"/>

        
        <param name="camera_rgb_topic" value="/$(arg camera)/rgb/image_raw"/>
        <param name="camera_depth_topic" value="/$(arg camera)/depth/image_raw"/>

	
	<param name="base_frame_name" value="odom"/>
	<param name="camera_frame_name" value="camera_rgb_optical_frame"/>
        
        
        <param name="point_cloud_downscale" value="2"/>

        
        <param name="fx" value="540.175979"/>
        <param name="fy" value="530.740137"/>
        <param name="cx" value="325.399891"/>
        <param name="cy" value="239.056878"/>
        <param name="rows" value="480"/>
        <param name="cols" value="640"/>
        <param name="camera_distance_max" value="3"/>
        <param name="camera_distance_min" value="0.45"/>
        
        
        <param name="base_frame_name" value="odom"/>

        <param name="map_resolution" value="0.05"/>
        <param name="min_voxel_weight" value="100"/>
        <param name="ground_level" value="0.05"/>
        <param name="agent_height" value="1.5"/>
        <param name="enable_chisel" value="false"/>
        <param name="chisel_step" value="30"/>
        <param name="height_color" value="true"/>

    </node>

    <node pkg="rviz" type="rviz" output="screen" name="rviz" args="-d $(find skimap_ros)/rviz/skimap_live.rviz" />

</launch>

m4nh, 2017-06-08:
@AmirCognitive hi! Maybe camera frame si wrong, can you send me a bag file with a ride of that robot?

m4nh, 2017-06-08:
@AmirCognitive  so please answer first the opened issues before opening other issues! Because i'm alone and is not simple for me to answer multiple questions in the same time! :D

AmirCognitive, 2017-06-09:
I understand, thanks for reply ;)

I tried different things for camera frame but none of them worked correctly, so uploaded the bag file in the following link (though its a bit big in size):

https://drive.google.com/open?id=0Bwsg0i6pm8bSblYwYzJRT19YS0E

YiChenCityU, 2018-04-25:
Hi @Amir-Ramezani It's good you have run the system.  Could you tell me why I cannot get the 3D map with tiago_lar.bag file? Thanks very much.
[image]

closed 2018-05-21 by m4nh