m4nh/skimap_ros #8: can't locate node [slamdunk_tracker] in package [skimap_ros]

Sangheli, 2017-09-01:
When I try launch slamdunk_tracker.launch, I have got error:

cannot launch node of type [skimap_ros/slamdunk_tracker]: can't locate node [slamdunk_tracker] in package [skimap_ros]

how can I fix that?

m4nh, 2017-09-01:
seems you try launch "rosrun" instead of "roslaunch" can you post your command?

ytgcljj, 2017-09-05:
cannot launch node of type [skimap_ros/slamdunk_tracker]: can't locate node [slamdunk_tracker] in package [skimap_ros]

how can I fix that?
I don't know how to run rosrun

m4nh, 2017-09-05:
i mean: write here the command that you launch not the answer of the system.

ytgcljj, 2017-09-06:
I run  roslaunch skimap_ros slamdunk_tracker.launch

m4nh, 2017-09-06:
yes you both have to enable the option "BUILD_SLAMDUNK" (ON) to build slamdunk libraries. If is there some error, can you paste the output here?

Sangheli, 2017-09-06:
I builed with "BUIL_SLAMDUNK" (ON), without erros, but when i use roslaunch skimap_ros slamdunk_tracker.launch

i got this:
ERROR: cannot launch node of type [skimap_ros/slamdunk_tracker]: can't locate node [slamdunk_tracker] in package [skimap_ros]

m4nh, 2017-09-06:
is very strange behaviour, i can't help you in this way. Can you paste the catkin_make whole output?

Sangheli, 2017-09-06:
Try build catkin second time, and i have errors during make:

Base path: /home/user/catkin_ws
Source space: /home/user/catkin_ws/src
Build space: /home/user/catkin_ws/build
Devel space: /home/user/catkin_ws/devel
Install space: /home/user/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/user/catkin_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/user/catkin_ws/build"
####
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_cpp
[  2%] Built target skimap_live
[  2%] Built target geometry_msgs_generate_messages_nodejs
[  2%] Built target geometry_msgs_generate_messages_py
[  2%] Built target _skimap_ros_generate_messages_check_deps_SkimapIntegrationService
[  2%] Built target std_msgs_generate_messages_nodejs
[  2%] Built target geometry_msgs_generate_messages_lisp
[  2%] Built target std_msgs_generate_messages_lisp
[  2%] Built target geometry_msgs_generate_messages_eus
[  2%] Built target std_msgs_generate_messages_eus
[  2%] Built target std_msgs_generate_messages_py
[  4%] Built target image_publisher_gencfg
[  5%] Built target image_view_gencfg
[  8%] Built target disparity_view
[ 10%] Built target stereo_view
[ 16%] Built target video_recorder
[ 16%] Built target extract_images
[ 21%] Built target image_view
[ 24%] Built target image_saver
[ 29%] Built target image_proc_gencfg
[ 32%] Built target image_publisher_exe
[ 33%] Built target image_rotate_gencfg
[ 35%] Built target stereo_image_proc_gencfg
[ 36%] Built target skimap_ros_generate_messages_cpp
[ 39%] Built target image_rotate_exe
[ 41%] Built target skimap_ros_generate_messages_py
[ 44%] Built target integration_of_random_points
[ 45%] Built target skimap_ros_generate_messages_nodejs
[ 47%] Built target skimap_ros_generate_messages_lisp
[ 50%] Built target skimap_ros_generate_messages_eus
[ 63%] Built target depth_image_proc
[ 66%] Built target skimap_ros
[ 68%] Built target image_publisher
[ 71%] Built target image_view_exe
[ 71%] Built target skimap_ros_generate_messages
[ 74%] Built target image_rotate
[ 77%] Built target skimap_map_service
[ 89%] Built target image_proc
[ 91%] Built target image_proc_exe
[ 97%] Built target stereo_image_proc
[100%] Built target stereoimageproc_exe
user@ubuntu-server:~/catkin_ws$ catkin_make
Base path: /home/user/catkin_ws
Source space: /home/user/catkin_ws/src
Build space: /home/user/catkin_ws/build
Devel space: /home/user/catkin_ws/devel
Install space: /home/user/catkin_ws/install
####
#### Running command: "cmake /home/user/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/user/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/user/catkin_ws/install -G Unix Makefiles" in "/home/user/catkin_ws/build"
####
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
[... 553 lines cut ...]
[ 86%] Built target stereoimageproc_exe
In file included from /home/user/catkin_ws/src/skimap_ros/src/slamdunk/feature_tracker.cpp:9:0:
/home/user/catkin_ws/src/skimap_ros/include/slamdunk/camera_tracker.h:20:42: fatal error: opencv2/nonfree/features2d.hpp: No such file or directory
compilation terminated.
skimap_ros/CMakeFiles/slamdunk.dir/build.make:110: recipe for target 'skimap_ros/CMakeFiles/slamdunk.dir/src/slamdunk/feature_tracker.cpp.o' failed
make[2]: *** [skimap_ros/CMakeFiles/slamdunk.dir/src/slamdunk/feature_tracker.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/user/catkin_ws/src/skimap_ros/include/slamdunk/slam_dunk.h:13:0,
                 from /home/user/catkin_ws/src/skimap_ros/src/slamdunk/slam_dunk.cpp:9:
/home/user/catkin_ws/src/skimap_ros/include/slamdunk/camera_tracker.h:20:42: fatal error: opencv2/nonfree/features2d.hpp: No such file or directory
compilation terminated.
skimap_ros/CMakeFiles/slamdunk.dir/build.make:206: recipe for target 'skimap_ros/CMakeFiles/slamdunk.dir/src/slamdunk/slam_dunk.cpp.o' failed
make[2]: *** [skimap_ros/CMakeFiles/slamdunk.dir/src/slamdunk/slam_dunk.cpp.o] Error 1
In file included from /home/user/catkin_ws/src/skimap_ros/include/slamdunk/slam_dunk.h:13:0,
                 from /home/user/catkin_ws/src/skimap_ros/include/slamdunk_extension/SlamDunkScene.h:12,
                 from /home/user/catkin_ws/src/skimap_ros/src/slamdunk_extension/SlamDunkScene.cpp:9:
/home/user/catkin_ws/src/skimap_ros/include/slamdunk/camera_tracker.h:20:42: fatal error: opencv2/nonfree/features2d.hpp: No such file or directory
compilation terminated.
skimap_ros/CMakeFiles/slamdunk.dir/build.make:254: recipe for target 'skimap_ros/CMakeFiles/slamdunk.dir/src/slamdunk_extension/SlamDunkScene.cpp.o' failed
make[2]: *** [skimap_ros/CMakeFiles/slamdunk.dir/src/slamdunk_extension/SlamDunkScene.cpp.o] Error 1
CMakeFiles/Makefile2:4230: recipe for target 'skimap_ros/CMakeFiles/slamdunk.dir/all' failed
make[1]: *** [skimap_ros/CMakeFiles/slamdunk.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

m4nh, 2017-09-06:
I see a "fatal error: opencv2/nonfree/features2d.hpp: No such file or directory" ... it is not a compilation "without error" ...

Sangheli, 2017-09-06:
Which version on opencv I should use? in system installed version 3.3.0, in ros kinetik - 3.2.0

m4nh, 2017-09-06:
No slamdunk is compatible only with Opencv2.4.. for this reason it was removed from the official distribution of Skimap.. in the Readme of the package there is an example (video) integrating Skimap with the new slam system ORBSLAM2.. maybe if you need a camera tracker is more simple to use it for now

Sangheli, 2017-09-06:
How i can connect orb_slam2 with skimap_ros?

m4nh, 2017-09-06:
sorry but i can't write a tutorial now :) but you can follow this issue:

https://github.com/m4nh/skimap_ros/issues/7

where we talked about it.
In the month of October we will release a plug&play package integrating ORBSLAM2 and SKIMAP, if you can wait

ytgcljj, 2017-09-07:
Base path: /home/gtx-16ros/dashgo_ws
Source space: /home/gtx-16ros/dashgo_ws/src
Build space: /home/gtx-16ros/dashgo_ws/build
Devel space: /home/gtx-16ros/dashgo_ws/devel
Install space: /home/gtx-16ros/dashgo_ws/install
####
#### Running command: "cmake /home/gtx-16ros/dashgo_ws/src -DCATKIN_DEVEL_PREFIX=/home/gtx-16ros/dashgo_ws/devel -DCMAKE_INSTALL_PREFIX=/home/gtx-16ros/dashgo_ws/install -G Unix Makefiles" in "/home/gtx-16ros/dashgo_ws/build"
####
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/gtx-16ros/dashgo_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/gtx-16ros/dashgo_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/gtx-16ros/dashgo_ws/devel;/opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/gtx-16ros/dashgo_ws/build/test_results
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.6
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 14 packages in topological order:
-- ~~  - hector_nav_msgs
-- ~~  - dashgo_description
-- ~~  - dashgo_driver
-- ~~  - dashgo_nav
-- ~~  - dashgo_rviz
-- ~~  - flashgo
-- ~~  - rplidar_ros
-- ~~  - simple_voice
-- ~~  - skimap_ros
-- ~~  - dashgo_tools
-- ~~  - pathgo_imu
-- ~~  - usb_cam
-- ~~  - hector_exploration_planner
-- ~~  - hector_exploration_node
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'hector_nav_msgs'
-- ==> add_subdirectory(dashgo/hector_nav_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- hector_nav_msgs: 0 messages, 5 services
-- +++ processing catkin package: 'dashgo_description'
-- ==> add_subdirectory(dashgo/dashgo_description)
-- +++ processing catkin package: 'dashgo_driver'
-- ==> add_subdirectory(dashgo/dashgo_driver)
[... 247 lines cut ...]
Scanning dependencies of target hector_nav_msgs_generate_messages
[ 86%] Built target hector_nav_msgs_generate_messages
Scanning dependencies of target simple_voice_generate_messages
[ 86%] Built target simple_voice_generate_messages
Scanning dependencies of target skimap_map_service
[ 88%] Building CXX object skimap_ros/CMakeFiles/skimap_map_service.dir/src/nodes/skimap_map_service.cpp.o
[ 89%] Building CXX object dashgo/hector_exploration_planner/CMakeFiles/hector_exploration_planner.dir/src/hector_exploration_planner.cpp.o
[ 90%] Linking CXX executable /home/gtx-16ros/dashgo_ws/devel/lib/usb_cam/usb_cam_node
[ 92%] Linking CXX executable /home/gtx-16ros/dashgo_ws/devel/lib/skimap_ros/skimap_live
[ 92%] Built target usb_cam_node
Scanning dependencies of target skimap_ros_generate_messages
[ 92%] Built target skimap_ros_generate_messages
[ 92%] Built target skimap_live
[ 93%] Linking CXX executable /home/gtx-16ros/dashgo_ws/devel/lib/skimap_ros/skimap_map_service
[ 93%] Built target skimap_map_service
[ 94%] Linking CXX shared library /home/gtx-16ros/dashgo_ws/devel/lib/libhector_exploration_planner.so
[ 94%] Built target hector_exploration_planner
Scanning dependencies of target hector_exploration_base_global_planner_plugin
[ 96%] Building CXX object dashgo/hector_exploration_planner/CMakeFiles/hector_exploration_base_global_planner_plugin.dir/src/hector_exploration_base_global_planner_plugin.cpp.o
[ 97%] Linking CXX shared library /home/gtx-16ros/dashgo_ws/devel/lib/libhector_exploration_base_global_planner_plugin.so
[ 97%] Built target hector_exploration_base_global_planner_plugin
Scanning dependencies of target exploration_planner_node
[ 98%] Building CXX object dashgo/hector_exploration_node/CMakeFiles/exploration_planner_node.dir/src/exploration_node.cpp.o
[100%] Linking CXX executable /home/gtx-16ros/dashgo_ws/devel/lib/hector_exploration_node/exploration_planner_node
[100%] Built target exploration_planner_node

ytgcljj, 2017-09-07:
setting /run_id to b9676810-93a8-11e7-8038-448a5bbfc220
process[rosout-1]: started with pid [7715]
started core service [/rosout]
ERROR: cannot launch node of type [skimap_ros/slamdunk_tracker]: can't locate node [slamdunk_tracker] in package [skimap_ros]
process[rviz-3]: started with pid [7732]
[ INFO] [1504773853.798546614]: rviz version 1.12.10
[ INFO] [1504773853.798588323]: compiled against Qt version 5.5.1
[ INFO] [1504773853.798599984]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1504773853.891227267]: Stereo is NOT SUPPORTED
[ INFO] [1504773853.891306340]: OpenGl version: 3 (GLSL 1.3).

m4nh, 2017-09-07:
@ytgcljj  I repost the same answer:
"""
sorry but i can't write a tutorial now :) but you can follow this issue:

#7

where we talked about it.
In the month of October we will release a plug&play package integrating ORBSLAM2 and SKIMAP, if you can wait
"""
Also you, as i can see, have the Opencv 3.2 version, which is not compatible with Slamdunk

closed 2018-05-21 by m4nh