m4nh/skimap_ros #9: Different timestamps inside tiago_lab.bag

AndreaIncertiDelmonte, 2017-10-04:
Hi,
running `rosbag play tiago_lar.bag --clock --pause` I noticed that there is a difference of 111.7 seconds between the bag time and messages timestamps.

This can be verified with the following python code:
```python
import rosbag
msg_counter = 1
for topic, msg, t in rosbag.Bag('tiago_lar.bag').read_messages():
    #if topic != "/tf":
    if topic == "/xtion/rgb/image_raw":
        print("Topic: {}".format(topic))
        print("t.to_sec(): {}".format(t.to_sec()))
        print("msg.header.stamp.to_sec(): {}".format(msg.header.stamp.to_sec()))
        print("Delta time: {}".format(t.to_sec() - (msg.header.stamp.to_sec())))
        print("")
        msg_counter = msg_counter + 1
    if msg_counter == 5:
        break
```
The code produce the following output:
```python
Topic: /xtion/rgb/image_raw
t.to_sec(): 1472902408.28
msg.header.stamp.to_sec(): 1472902296.56
Delta time: 111.716982841

Topic: /xtion/rgb/image_raw
t.to_sec(): 1472902408.37
msg.header.stamp.to_sec(): 1472902296.66
Delta time: 111.707210302

Topic: /xtion/rgb/image_raw
t.to_sec(): 1472902408.73
msg.header.stamp.to_sec(): 1472902297.0
Delta time: 111.730161905

Topic: /xtion/rgb/image_raw
t.to_sec(): 1472902408.87
msg.header.stamp.to_sec(): 1472902297.1
Delta time: 111.778051376

```
So in this situation when I publish the TF from orbcamera_camera to world (ORBSLAM node)  I have two secenarios;

- if I choose ros::Time::now() as timestap on Skimap side I get  Lookup would require extrapolation into the **past**.  Requested time 1472902308.293575041 but the earliest data is at time 1472902410.10, when looking up transform from frame [orbslam_camera] to frame [world].
- If I use insted msgRGB->header.stamp as timestamp I get Lookup would require extrapolation into the **future**.  Requested time 1472902309.564114801 but the latest data is at time 1472902309.430578701, when looking up transform from frame [orbslam_camera] to frame [world].

Do you have any hint to fix this issue?

Thanks
Andrea

m4nh, 2017-10-04:
@AndreaIncertiDelmonte  have you set the parameter:

rosparam set use_sim_time true

To force the sim time equal to one in the bag?
It's strange behaviour  .. however you can try to use as time "-1" or ros::Time(0) to pick the first available TF in the tree when you use tf_listener

m4nh, 2017-10-04:
@AndreaIncertiDelmonte  comunque io lavoro all'università di Bologna :) vedo che tu sei di reggio emilia.. Se ti interessa qualcosa in particolare su Skiamp puoi scrivermi in privato perché invece a me interessa qualcuno che lavori sull'integrazione con ORBSLAM2. Fammi sapere