m4nh/skimap_ros #10: compilation issue:

rnunziata, 2017-10-05:
I am using opencv 3.2.0-dev

```
In file included from /home/rjn/catkin_ws/src/skimap_ros/include/slamdunk/camera_tracker.h:17:0,
                 from /home/rjn/catkin_ws/src/skimap_ros/src/slamdunk/feature_tracker.cpp:9:
/home/rjn/catkin_ws/src/skimap_ros/include/slamdunk/quadtree.h: In function ‘cv::Mat slamdunk::drawQuadTree(const QuadTreeT&, unsigned int)’:
/home/rjn/catkin_ws/src/skimap_ros/include/slamdunk/quadtree.h:332:5: error: ‘circle’ is not a member of ‘cv’
     cv::circle(img, cv::Point((it->m_coords.x() - tree.getOffset().x()) * imgsize / tree.getLength(), (it->m_coords.y() - tree.getOffset().y()) * imgsize / tree.getLength()),

```
and 

```
In file included from /home/rjn/catkin_ws/src/skimap_ros/include/slamdunk/camera_tracker.h:20:0,
                 from /home/rjn/catkin_ws/src/skimap_ros/src/slamdunk/feature_tracker.cpp:9:
/usr/include/opencv2/nonfree/features2d.hpp: At global scope:
/usr/include/opencv2/nonfree/features2d.hpp:73:21: error: ‘vector’ has not been declared
                     vector<KeyPoint>& keypoints) const;
                     ^
/usr/include/opencv2/nonfree/features2d.hpp:73:27: error: expected ‘,’ or ‘...’ before ‘<’ token
                     vector<KeyPoint>& keypoints) const;
                           ^
/usr/include/opencv2/nonfree/features2d.hpp:77:21: error: ‘vector’ has not been declared
                     vector<KeyPoint>& keypoints,
                     ^
/usr/include/opencv2/nonfree/features2d.hpp:77:27: error: expected ‘,’ or ‘...’ before ‘<’ token
                     vector<KeyPoint>& keypoints,
                           ^
/usr/include/opencv2/nonfree/features2d.hpp:76:10: error: ‘void cv::SIFT::operator()(cv::InputArray, cv::InputArray, int) const’ cannot be overloaded
     void operator()(InputArray img, InputArray mask,

```

m4nh, 2017-10-05:
@rnunziata  Sorry but unfortunately SlamDUnk is not compatible for now with OpenCV3.. you can use SkiMap however. What do you need to do?

rnunziata, 2017-10-05:
could I compile against a diff ver of opencv by point to it like this in the make file, any versions older then 2.4 I can use. I want to track without pose if I do not send pose data via rosbag then the map does not develop correctly . I am using a filtered version of rgbd_dataset_freiburg3_long_office_household_validation.bag with only depth, image, cam_info.

```
# OpenCV 2.4.9
CPPFLAGS = -L/path/to/opencv2.4.9/lib \
       -I/path/to/opencv2.4.9/include
```

m4nh, 2017-10-09:
@rnunziata  did you remove the "find(opencv..." row or not? in this case are you sure that your forced SET over Opencv_VARIABLES is after the "find" operation? 
By the way try to use ORBSLAM2 as camera tracker (See https://github.com/m4nh/skimap_ros/issues/7) because unfortunately SlamDUNK is not mantained continuously for now in this package, that is focused only on the SkiMap mapping part.

rnunziata, 2017-10-09:
This worked for me 

```
find_package(OpenCV 2.4.13.3 EXACT NO_DEFAULT_PATH REQUIRED PATHS /home/rjn/opencv-2.4.13.3/install)
if(OpenCV_INCLUDE_DIRS)
  include_directories(${OpenCV_INCLUDE_DIRS})
endif(OpenCV_INCLUDE_DIRS)

```

...but now I am getting error related to nonfree:  is this version no good?

```
In file included from /home/rjn/catkin_ws/src/skimap_ros/include/slamdunk/camera_tracker.h:20:0,
                 from /home/rjn/catkin_ws/src/skimap_ros/src/slamdunk/feature_tracker.cpp:9:
/home/rjn/opencv-2.4.13.3/install/include/opencv2/nonfree/features2d.hpp: At global scope:
/home/rjn/opencv-2.4.13.3/install/include/opencv2/nonfree/features2d.hpp:73:21: error: ‘vector’ has not been declared
                     vector<KeyPoint>& keypoints) const;
                     ^
/home/rjn/opencv-2.4.13.3/install/include/opencv2/nonfree/features2d.hpp:73:27: error: expected ‘,’ or ‘...’ before ‘<’ token
                     vector<KeyPoint>& keypoints) const;
                           ^
/home/rjn/opencv-2.4.13.3/install/include/opencv2/nonfree/features2d.hpp:77:21: error: ‘vector’ has not been declared
                     vector<KeyPoint>& keypoints,

```

rnunziata, 2017-10-09:
Have you looked at ros ptam for tf and pose transform ....

rnunziata, 2017-10-11:
Slamdunk does not compile with opencv-2.4.13.3.  I will return to 3.2 and try ORBSLAM2 for interface.

closed 2017-10-11 by rnunziata