m4nh/skimap_ros #11: Can skimap be intergrated with PTAM

rnunziata, 2017-10-15:
Can skimap be integrated with PTAM's published pose. By creating a node that takes the ptam pose and resend it as tf.  I am currently having issues getting PTAM to compile/run on my system , I will try this when I get it resolved but I wanted to know if there any issues that would stop this from working.

m4nh, 2017-10-15:
@rnunziata please don't spam issue tracker :) this is only a question! Please send your question by email tuo me. Howevre yes, The tracker doesnt matter, it will work also with ptam estimated pose.

closed 2017-10-15 by m4nh

rnunziata, 2017-10-18:
Sorry....I did do an integration with PTAM and posted it in issues....if this is not correct place . Please tell me where to post or should I email to you.  I also had issue with explained in post...I need to extract it out and post separately.  Sorry for miss using site.