m4nh/skimap_ros #13: If depth map is not in meters but mm

rnunziata, 2017-10-22:
If depth map is not in meters but mm say 5000 to a meter.
Is there anyway for skip map to work with these metrics.

m4nh, 2017-10-22:
It s a good idea to transform measurements in meters because the resolution of a large scale map in millimeters leads to a very huge Memory footprint.

rnunziata, 2017-10-22:
thanks. 
I modified a line 498  in skimap_live to adj to camera measure using parameter....is this a good way to do this?
depth_p[j] = float(depth_mm_p[j])/mapParameters.depthMapFactor;

m4nh, 2017-10-24:
@rnunziata i think its ok... but  is to remind you that the "skimap_live" is just an example node that we made for testing purposes; instead in the simple [Tutorial](https://github.com/m4nh/skimap_ros/blob/master/src/nodes/tutorials/integration_of_random_points.cpp) was explained how to use SkiMap with your custom software (ROS or ROS not).. so if you want to use Skimap with your custom business logic ("unfortunatelly") you have to design your own software.. So modifying "skimap_live" is good but is not the best solution for you :)

closed 2018-05-21 by m4nh