m4nh/skimap_ros #16: Can't build skimap_ros on ubuntu 14.04 indigo

teduav, 2017-12-08:
Hi,
      Thank you for your package.
      I want to build the Skimap_ros on ubuntu 14.04 indigo, the errors are below

       [100%] Building CXX object skimap_ros/CMakeFiles/skimap_map_service.dir/src/nodes/skimap_map_service.cpp.o
[100%] Built target skimap_ros_generate_messages
/home/exbot/catkin_ws/src/skimap_ros/src/nodes/skimap_live.cpp: In function ‘int main(int, char**)’:
/home/exbot/catkin_ws/src/skimap_ros/src/nodes/skimap_live.cpp:554:108: error: no matching function for call to ‘ros::NodeHandle::param(const char [26], const char [11])’
     std::string map_cloud_publisher_topic = nh->param<std::string>("map_cloud_publisher_topic","live_cloud");

m4nh, 2017-12-08:
@teduav  Can you post all the catkin_make log? because here appears to be cut at the end. However, did you modify the original code?

Xl-wj, 2018-03-15:
Hi, 
I am goint to build the Skimap_ros on ubuntu 14.04 indigo.
the errors are below:

Base path: /home/xl/catkin_ws
Source space: /home/xl/catkin_ws/src
Build space: /home/xl/catkin_ws/build
Devel space: /home/xl/catkin_ws/devel
Install space: /home/xl/catkin_ws/install
####
#### Running command: "cmake /home/xl/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/xl/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/xl/catkin_ws/install -G Unix Makefiles" in "/home/xl/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/xl/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/xl/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.19
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - skimap_ros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'skimap_ros'
-- ==> add_subdirectory(skimap_ros)
-- Using these message generators: gencpp;genlisp;genpy
-- skimap_ros: 0 messages, 1 services
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Found OpenMP: -fopenmp  
-- Found OpenMP
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "2.91.0") 
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   random
--   timer
--   system
-- Configuring done
-- Generating done
-- Build files have been written to: /home/xl/catkin_ws/build
####
#### Running command: "make -j8 -l8" in "/home/xl/catkin_ws/build"
####
Scanning dependencies of target std_msgs_generate_messages_cpp
Scanning dependencies of target geometry_msgs_generate_messages_cpp
Scanning dependencies of target skimap_live
Scanning dependencies of target skimap_ros
Scanning dependencies of target std_msgs_generate_messages_py
Scanning dependencies of target geometry_msgs_generate_messages_py
Scanning dependencies of target _skimap_ros_generate_messages_check_deps_SkimapIntegrationService
Scanning dependencies of target integration_of_random_points
[  0%] [  0%] Built target std_msgs_generate_messages_cpp
[  0%] [  0%] Built target std_msgs_generate_messages_py
Built target geometry_msgs_generate_messages_py
Built target geometry_msgs_generate_messages_cpp
Scanning dependencies of target std_msgs_generate_messages_lisp
Scanning dependencies of target geometry_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target _skimap_ros_generate_messages_check_deps_SkimapIntegrationService
[... 18 lines cut ...]
                 from /home/xl/catkin_ws/src/skimap_ros/include/skimap/SkiMap.hpp:16,
                 from /home/xl/catkin_ws/src/skimap_ros/src/nodes/tutorials/integration_of_random_points.cpp:17:
/home/xl/catkin_ws/src/skimap_ros/include/skimap/voxels/GenericVoxel3D.hpp: In function ‘std::ostream& skimap::operator<<(std::ostream&, const skimap::GenericVoxel3D<V, D>&)’:
/home/xl/catkin_ws/src/skimap_ros/include/skimap/voxels/GenericVoxel3D.hpp:56:15: error: ‘setprecision’ is not a member of ‘std’
         os << std::setprecision(std::numeric_limits<double>::digits10 + 2);
               ^
In file included from /home/xl/catkin_ws/src/skimap_ros/include/skimap/SkiMap.hpp:16:0,
                 from /home/xl/catkin_ws/src/skimap_ros/src/nodes/tutorials/integration_of_random_points.cpp:17:
/home/xl/catkin_ws/src/skimap_ros/include/skimap/voxels/GenericTile2D.hpp: In function ‘std::ostream& skimap::operator<<(std::ostream&, const skimap::GenericTile2D<V, D>&)’:
/home/xl/catkin_ws/src/skimap_ros/include/skimap/voxels/GenericTile2D.hpp:48:15: error: ‘setprecision’ is not a member of ‘std’
         os << std::setprecision(std::numeric_limits<double>::digits10 + 2);
               ^
Linking CXX shared library /home/xl/catkin_ws/devel/lib/libskimap_ros.so
[100%] Built target skimap_ros
make[2]: *** [skimap_ros/CMakeFiles/integration_of_random_points.dir/src/nodes/tutorials/integration_of_random_points.cpp.o] 错误 1
make[1]: *** [skimap_ros/CMakeFiles/integration_of_random_points.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务....
Linking CXX executable /home/xl/catkin_ws/devel/lib/skimap_ros/skimap_map_service
[100%] Built target skimap_map_service
Linking CXX executable /home/xl/catkin_ws/devel/lib/skimap_ros/skimap_live
[100%] Built target skimap_live
make: *** [all] 错误 2
Invoking "make -j8 -l8" failed
xl@xl:~/catkin_ws$ ./devel/lib/skimap_ros/skimap_
skimap_live         skimap_map_service

Xl-wj, 2018-03-15:
Hi,
I am going to build the Skimap_ros on ubuntu 14.04 indigo.
 however, the errors are below,

Base path: /home/xl/catkin_ws
Source space: /home/xl/catkin_ws/src
Build space: /home/xl/catkin_ws/build
Devel space: /home/xl/catkin_ws/devel
Install space: /home/xl/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/xl/catkin_ws/build"
####
####
#### Running command: "make clean -j8 -l8" in "/home/xl/catkin_ws/build"
####
xl@xl:~/catkin_ws$ catkin_make
Base path: /home/xl/catkin_ws
Source space: /home/xl/catkin_ws/src
Build space: /home/xl/catkin_ws/build
Devel space: /home/xl/catkin_ws/devel
Install space: /home/xl/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/xl/catkin_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/xl/catkin_ws/build"
####
[  0%] [  0%] Built target std_msgs_generate_messages_py
[  0%] [  0%] Built target geometry_msgs_generate_messages_cpp
Built target std_msgs_generate_messages_cpp
Built target geometry_msgs_generate_messages_py
[  0%] [ 12%] Built target std_msgs_generate_messages_lisp
[ 12%] Built target geometry_msgs_generate_messages_lisp
Building CXX object skimap_ros/CMakeFiles/skimap_ros.dir/src/skimap_ros/skimap_ros_void_library.cpp.o
[ 12%] Built target _skimap_ros_generate_messages_check_deps_SkimapIntegrationService
[ 25%] Building CXX object skimap_ros/CMakeFiles/integration_of_random_points.dir/src/nodes/tutorials/integration_of_random_points.cpp.o
[ 37%] [ 50%] Generating C++ code from skimap_ros/SkimapIntegrationService.srv
Generating Python code from SRV skimap_ros/SkimapIntegrationService
[ 62%] Generating Lisp code from skimap_ros/SkimapIntegrationService.srv
[ 75%] Building CXX object skimap_ros/CMakeFiles/skimap_live.dir/src/nodes/skimap_live.cpp.o
[ 75%] [ 87%] Built target skimap_ros_generate_messages_lisp
Generating Python srv __init__.py for skimap_ros
[ 87%] Built target skimap_ros_generate_messages_py
[ 87%] Built target skimap_ros_generate_messages_cpp
[ 87%] Built target skimap_ros_generate_messages
Scanning dependencies of target skimap_map_service
[100%] Building CXX object skimap_ros/CMakeFiles/skimap_map_service.dir/src/nodes/skimap_map_service.cpp.o
In file included from /home/xl/catkin_ws/src/skimap_ros/include/skimap/voxels/GenericTile2D.hpp:11:0,
                 from /home/xl/catkin_ws/src/skimap_ros/include/skimap/SkiMap.hpp:16,
                 from /home/xl/catkin_ws/src/skimap_ros/src/nodes/tutorials/integration_of_random_points.cpp:17:
/home/xl/catkin_ws/src/skimap_ros/include/skimap/voxels/GenericVoxel3D.hpp: In function ‘std::ostream& skimap::operator<<(std::ostream&, const skimap::GenericVoxel3D<V, D>&)’:
/home/xl/catkin_ws/src/skimap_ros/include/skimap/voxels/GenericVoxel3D.hpp:56:15: error: ‘setprecision’ is not a member of ‘std’
         os << std::setprecision(std::numeric_limits<double>::digits10 + 2);
               ^
In file included from /home/xl/catkin_ws/src/skimap_ros/include/skimap/SkiMap.hpp:16:0,
                 from /home/xl/catkin_ws/src/skimap_ros/src/nodes/tutorials/integration_of_random_points.cpp:17:
/home/xl/catkin_ws/src/skimap_ros/include/skimap/voxels/GenericTile2D.hpp: In function ‘std::ostream& skimap::operator<<(std::ostream&, const skimap::GenericTile2D<V, D>&)’:
/home/xl/catkin_ws/src/skimap_ros/include/skimap/voxels/GenericTile2D.hpp:48:15: error: ‘setprecision’ is not a member of ‘std’
         os << std::setprecision(std::numeric_limits<double>::digits10 + 2);
               ^
Linking CXX shared library /home/xl/catkin_ws/devel/lib/libskimap_ros.so
[100%] Built target skimap_ros
make[2]: *** [skimap_ros/CMakeFiles/integration_of_random_points.dir/src/nodes/tutorials/integration_of_random_points.cpp.o] error 1
make[1]: *** [skimap_ros/CMakeFiles/integration_of_random_points.dir/all] error 2
make[1]: *** waiting....
Linking CXX executable /home/xl/catkin_ws/devel/lib/skimap_ros/skimap_map_service
[100%] Built target skimap_map_service
Linking CXX executable /home/xl/catkin_ws/devel/lib/skimap_ros/skimap_live
[100%] Built target skimap_live
make: *** [all] error 2
Invoking "make -j8 -l8" failed

m4nh, 2018-03-15:
Can you run the single thread procedure with "catkin_make -j1" ? In this way the output log is more clear for me, thanks

Xl-wj, 2018-03-16:
Thanks for your reply.
I compile with "catkin_make -j1",
Message as follows：

Base path: /home/xl/catkin_ws
Source space: /home/xl/catkin_ws/src
Build space: /home/xl/catkin_ws/build
Devel space: /home/xl/catkin_ws/devel
Install space: /home/xl/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/xl/catkin_ws/build"
####
####
#### Running command: "make -j1" in "/home/xl/catkin_ws/build"
####
[ 12%] Building CXX object skimap_ros/CMakeFiles/integration_of_random_points.dir/src/nodes/tutorials/integration_of_random_points.cpp.o
In file included from /home/xl/catkin_ws/src/skimap_ros/include/skimap/voxels/GenericTile2D.hpp:11:0,
                 from /home/xl/catkin_ws/src/skimap_ros/include/skimap/SkiMap.hpp:16,
                 from /home/xl/catkin_ws/src/skimap_ros/src/nodes/tutorials/integration_of_random_points.cpp:17:
/home/xl/catkin_ws/src/skimap_ros/include/skimap/voxels/GenericVoxel3D.hpp: In function ‘std::ostream& skimap::operator<<(std::ostream&, const skimap::GenericVoxel3D<V, D>&)’:
/home/xl/catkin_ws/src/skimap_ros/include/skimap/voxels/GenericVoxel3D.hpp:56:15: error: ‘setprecision’ is not a member of ‘std’
         os << std::setprecision(std::numeric_limits<double>::digits10 + 2);
               ^
In file included from /home/xl/catkin_ws/src/skimap_ros/include/skimap/SkiMap.hpp:16:0,
                 from /home/xl/catkin_ws/src/skimap_ros/src/nodes/tutorials/integration_of_random_points.cpp:17:
/home/xl/catkin_ws/src/skimap_ros/include/skimap/voxels/GenericTile2D.hpp: In function ‘std::ostream& skimap::operator<<(std::ostream&, const skimap::GenericTile2D<V, D>&)’:
/home/xl/catkin_ws/src/skimap_ros/include/skimap/voxels/GenericTile2D.hpp:48:15: error: ‘setprecision’ is not a member of ‘std’
         os << std::setprecision(std::numeric_limits<double>::digits10 + 2);
               ^
make[2]: *** [skimap_ros/CMakeFiles/integration_of_random_points.dir/src/nodes/tutorials/integration_of_random_points.cpp.o] error 1
make[1]: *** [skimap_ros/CMakeFiles/integration_of_random_points.dir/all] error 2
make: *** [all] error 2
Invoking "make -j1" failed

m4nh, 2018-03-16:
@Xl-wj  which is the version of your "gcc" compiler? Check it with "gcc --version" .. however i think that you have to update the version at least to the 5.4

Xl-wj, 2018-03-19:
Thanks you for your help.
I have solved this problem.
I add a header "#include <<iomanip>> " in the file: catkin_ws/src/skimap_ros/include/skimap/voxels/GenericVoxel3D.hpp.

Xl-wj, 2018-03-19:
Thanks you for your help.
I have solved this problem.
I add a header "#include <iomanip> " in the file: catkin_ws/src/skimap_ros/include/skimap/voxels/GenericVoxel3D.hpp.

m4nh, 2018-03-19:
@Xl-wj  which header? It's useful to share your solution because we can implement it in the main repository to solve the same problem for other users! :) thanks

Xl-wj, 2018-03-19:
It's "#include <iomanip>"  in the file: catkin_ws/src/skimap_ros/include/skimap/voxels/GenericVoxel3D.hpp.

Sorry for not copying the line of code last time.

Xl-wj, 2018-03-19:
It's "#include '<iomanip>' " in the file: catkin_ws/src/skimap_ros/include/skimap/voxels/GenericVoxel3D.hpp.

Sorry for not copying the line of code last time.

Xl-wj, 2018-03-19:
It's "#include iomanip" in the file: catkin_ws/src/skimap_ros/include/skimap/voxels/GenericVoxel3D.hpp.

Sorry for not copying the line of code last time.

closed 2018-05-21 by m4nh