m4nh/skimap_ros #17: roslaunch skimap_ros slamdunk_tracker.launch

liyi2017, 2018-04-12:
Hi! I'm trying to slamdunk_tracker.launch,I run this :
1) roslaunch skimap_ros slamdunk_tracker.launch
2) rosbag play tiago_lar.bag --clock
However, I got the error :1、"Fixed Frame [slam_map] does not exist" on the RVIZ
2、“ERROR: cannot launch node of type [skimap_ros/slamdunk_tracker]: can't locate node [slamdunk_tracker] in package [skimap_ros]”after the slamdunk_tracker.launch
How can I solve this problem?thx
i catkin_make the ws is not error

m4nh, 2018-04-12:
@liyi2017  did you launch "source devel/setup.bash" in the ws folder?

liyi2017, 2018-04-15:
yes , i did"source catkin_ws/devel/setup.bash". but not use.
and the rviz  Fixed Frame have a error "No tf data.  Actual error: Fixed Frame [slam_map] does not exist"
thank you for your help.

closed 2018-05-21 by m4nh