m4nh/skimap_ros #20: error about slamdunk_tracker.launch

YiChenCityU, 2018-04-26:
Hi,
I also found there is an error when run this command : roslaunch skimap_ros slamdunk_tracker.launch 
ERROR: cannot launch node of type [skimap_ros/slamdunk_tracker]: can't locate node [slamdunk_tracker] in package [skimap_ros]

I found that in CmakeList.txt there is no add_executable about slamdunk_tracker.cpp
```
add_executable(skimap_live src/nodes/skimap_live.cpp)
target_link_libraries(skimap_live ${OpenCV_LIBRARIES} ${catkin_LIBRARIES})

add_executable(skimap_map_service src/nodes/skimap_map_service.cpp)
target_link_libraries(skimap_map_service ${OpenCV_LIBRARIES} ${catkin_LIBRARIES})
add_dependencies(skimap_map_service skimap_ros_generate_messages_cpp)
```
What do you think?

m4nh, 2018-04-27:
Hi @YiChenCityU  i'm sorry but the SlamDunk Tracker is not supported anymore... i suggest to use OrbSlam2 as shown in the last video

m4nh, 2018-04-27:
@YiChenCityU  i corrected the Readme now... if you want to test the system, you can download the BAG as stated at the beginning of the Readme.. then launch:

> 1) roslaunch skimap_ros skimap_live.launch

> 2) rosbag play tiago_lar.bag --clock

closed 2018-05-21 by m4nh