m4nh/skimap_ros #21: 3D mapping result is weired

Bzhnja, 2018-05-17:
Hi @m4nh ,
I have tried this package on my own slam system, but the 3d mapping result is not correct, there are so many vertical strips, the result is as following,
[image]
and the LiveCloud topic is this, it is only several narrow vertical strips:
[image]
But actually my depth image looks like the image below.
[image]
This is the skimap_live.launch that I have modified.
`<launch>

    <arg name="camera" default="camera" />

    <node name="skimap_live" output="screen" pkg="skimap_ros" type="skimap_live">

        
        <param name="hz" value="10" />

        
        <param name="camera_rgb_topic" value="/$(arg camera)/rgb/image_raw" />
        <param name="camera_depth_topic" value="/$(arg camera)/depth_registered/image_raw" />

        
        <param name="point_cloud_downscale" value="1" />

        
        <param name="fx" value="520.5687" />
        <param name="fy" value="520.9826" />
        <param name="cx" value="328.6763" />
        <param name="cy" value="236.1047" />
        <param name="rows" value="480" />
        <param name="cols" value="640" />
        <param name="camera_distance_max" value="3" />
        <param name="camera_distance_min" value="0.45" />

        
        <param name="base_frame_name" value="vodom" />
        <param name="camera_frame_name" value="camera_rgb_optical_frame" />
        <param name="map_resolution" value="0.05" />
        <param name="min_voxel_weight" value="100" />
        <param name="ground_level" value="0.05" />
        <param name="agent_height" value="1.5" />
        <param name="enable_chisel" value="false" />
        <param name="chisel_step" value="30" />
        <param name="height_color" value="true" />

    </node>

    <node pkg="rviz" type="rviz" output="screen" name="rviz" args="-d $(find skimap_ros)/rviz/skimap_live.rviz" />

</launch>`
What's wrong with my operation, any suggestions? Thank you.

m4nh, 2018-05-17:
@Bzhnja hi! Maybe try to increase "camera_distance_max" from 3 to 10... maybe it's a FAR_CLIP_PLANE problem

Bzhnja, 2018-05-19:
@m4nh hi,
I have changed "camera_distance_max" to be 10, but the LiveCloud topic is still not correct.

[image]

m4nh, 2018-05-21:
@Bzhnja what's your RGB-D sensor? are you publishing DEPTH data by yourself or by means of the official driver?