m4nh/skimap_ros #22: Local_robot_map

koko19, 2018-05-18:
First of all, thank you very much for making this code public.

I have a question about Skimap functions. We are trying to build local map of the robot environment equipped with RGBD camera. For now we build octomap, then we extract the bounding box and afterwards add new point cloud to the last extracted bounding box. This process repeats continuously. Although the results are OK, the process is fairly slow and octomap structure does not allow for easy efficient implementation of this solution.

Could you please tell me if this is possible with Skimap. So, we want to build a map, crop map to remove everything outside a specified box and then update remaining box with new point cloud?

m4nh, 2018-05-18:
hi @koko19 ! Despite the map you use ( Octomap or Skimap) the best solution is to filter a priori your points setting max_x/min_x max_y/min_y and so on... however if you want to implement it in SkiMap, it has the SearchInRange function to filter points in a box.. but however it is a slow procedure..

koko19, 2018-05-18:
Thank you very much for qucik answer.

We do perform point cloud filtering before insertion, however as the robot moves the octomap grows. We do not need entire octomap just the fixed field around the robot. This is why we extract the box from the octomap and delete the rest. Once the new point cloud is inserted into the filtered part we repeat the procedure. This way we keep probabilistic values of area around the robot. The problem with octomap is that it is now built for these things and implementantion is difficult and slow. 

I was wondering if this could be done mor easily using Skimap?

m4nh, 2018-05-18:
@koko19  sorry but i can't help you with this feature because is not very clear for me :) it is an unusual way to perform mapping (and/or tracking) because the robot is moving so the portion of the space "around" the robot changes during time.. so i do not understood very well the problem, sorry

closed 2018-05-21 by m4nh