m4nh/skimap_ros #24: Clear free space

dario-ljubic, 2018-05-22:
I have managed to implement your mapping solution with our SLAM system, however I have a question regarding the free space management used by the SkiMap.

 When mapping areas with moving objects once the moving object is inserted into the SkiMap it is never filtered out, even when the same area is mapped without the object . By going through the code I was unable to find any ray casting implementation which would ensure that all voxels between the laser and the measured point are freed. Can you please confirm this and if so do you plan to release a version with this option, or do I have to implement it by myself. If this is the case, do you have any starting points?

m4nh, 2018-05-22:
Hi @dario-ljubic ! Thanks for the question.. Unfortunately there is no RayCasting solution (like Octomap) implemented in SkiMap to speedup the integration part. However you can implement the "de-integration" process to remove some measurement from the map. When you integrate new voxels you can see that there is the "weight" field of the voxel set to "1.0"; if you set the weight to "-1.0" the corresponding voxel will be removed (or eroded). With this technique we implemented the LOOP CLOSURE for example in OrbSlam2. In your case is more difficult because you do not have a measure to "de-integrate" but a generic random obstacle. 

However if you want you can implement the raycast procedure: for each point **p** you want to integrate you have to generate **n** points interpolated between **p** and the camera origin with an interpolation step equal to the voxel size (or even the half of it). 

Another simple solution is to implement a threshold on the "weight" field when you perform the visit of the map (like in the **skimap_live** node). In this way you are sure that the Voxel associated with a moving obstacle has a weight smaller than a fixed obstacle because was hit less. We successfully use this procedure also in the automotive environments where we deal with moving cars.

dario-ljubic, 2018-06-05:
Thank You for the quick answer!