m4nh/skimap_ros #27: SkiMap integration with Velodyne Lidar

BrunoEduardoCSantos, 2018-08-30:
Hi,
Is there an integration with velodyne lidar? Or just with RGB-D cameras?
Thanks in advance,
Bruno

m4nh, 2018-08-31:
hi @BrunoEduardoCSantos, SkiMap works with any type of 3D data, given that is a generic 3D Indexing Data Structure. In the [Basi Tutorial](https://github.com/m4nh/skimap_ros/blob/master/src/nodes/tutorials/integration_of_random_points.cpp) is explained how to integrate a generic 3D information in SkiMap (without a direct ROS integration).

BrunoEduardoCSantos, 2018-08-31:
Thanks @m4nh for your feedback . 
Once I subscribe a topic from the registered point cloud coming from a SLAM algorithm, I can use this basic tutorial to integrate as SkiMap, right?

m4nh, 2018-08-31:
@BrunoEduardoCSantos  yes sure

BrunoEduardoCSantos, 2018-09-28:
Hi @BrunoEduardoCSantos ,
Currently, I am trying to get skimap viz on rviz using as example you skipmap_live. 
Although, instead rgb-d data, I am using PointXYZI. After getting the visualization in rviz the voxel colors were a mixed of colors which doesnt make sense at all.
My code to integrate the point cloud is the following:
`// integrate point cloud
  for(int i=0; i<pc->points.size();i++){
     double x = pc->points[i].x;
     double y = pc->points[i].y;
     double z = pc->points[i].z;
      //store voxel
      VoxelDataColor voxel;
      voxel.r = 0.1;
      voxel.g = 0.1;
      voxel.b = 0.1;
      voxel.w = 1.0;
      /**
       * Integration of custom voxel in the SkiMap data structure.
       */
      map_->integrateVoxel(CoordinateType(x), CoordinateType(y),
    CoordinateType(z), &voxel);
  }
`
Is there anyway of doing using intensity to assign the voxel color?
Regards,
Bruno

closed 2018-09-28 by BrunoEduardoCSantos

m4nh, 2018-09-28:
hi @BrunoEduardoCSantos ! can you send me a screeshot of the visualization?

BrunoEduardoCSantos, 2018-09-28:
Hi @m4nh ,
Thanks for the quick answer.
Here it goes.
[image]

BrunoEduardoCSantos, 2018-09-28:
[image]

BrunoEduardoCSantos, 2018-09-28:
I create my own node . I am not using really the skimap_live. Just a version of it. 
```
void SkimapPathPlanning::fillVisualizationMarkerWithVoxels(
    visualization_msgs::Marker &voxels_marker, std::vector<Voxel3D> &voxels) {

  cv::Mat colorSpace(1, voxels.size(), CV_32FC3);
//   if (mapParameters.height_color) {
    for (int i = 0; i < voxels.size(); i++) {
      colorSpace.at<cv::Vec3f>(i)[0] = 180 - (voxels[i].z / 2) * 180;
      colorSpace.at<cv::Vec3f>(i)[1] = 1;
      colorSpace.at<cv::Vec3f>(i)[2] = 1;
    }
    cv::cvtColor(colorSpace, colorSpace, CV_HSV2BGR);
//   }

  for (int i = 0; i < voxels.size(); i++) {

    /**
     * Create 3D Point from 3D Voxel
     */
    geometry_msgs::Point point;
    point.x = voxels[i].x;
    point.y = voxels[i].y;
    point.z = voxels[i].z;

    /**
     * Assign Cube Color from Voxel Color
     */
    std_msgs::ColorRGBA color;
    // if (mapParameters.height_color) {
      color.r = colorSpace.at<cv::Vec3f>(i)[2];
      color.g = colorSpace.at<cv::Vec3f>(i)[1];
      color.b = colorSpace.at<cv::Vec3f>(i)[0];
    // } else {
    //   color.r = float(voxels[i].data->r) / 255.0;
    //   color.g = float(voxels[i].data->g) / 255.0;
    //   color.b = float(voxels[i].data->b) / 255.0;
    // }
    color.a = 1;

    voxels_marker.points.push_back(point);
    voxels_marker.colors.push_back(color);
  }
}

```
Since I want a height map, is this the way to do it?
Thanks again.

BrunoEduardoCSantos, 2018-09-28:
It worked just replacing by $.y$.  Thanks for your help!

BrunoEduardoCSantos, 2018-09-28:
But it still has some noisy data.
[image]

BrunoEduardoCSantos, 2018-09-28:
by noisy I mean non related color data.

m4nh, 2018-09-28:
@BrunoEduardoCSantos .. sorry but the skimap_live is just a prototype for demo purposes :) ... however when you set:

```
colorSpace.at<cv::Vec3f>(i)[0] = 180 - (voxels[i].y / 2) * 180;
```

you are codifying the HUE of the color.. in this equation the max allowed y is obviously between [0,2].. probably negative values are creating noisy data... try to change this function for your custom purposes.. maybe managing negative numbers, changing the maximum range (>2) and so on..

BrunoEduardoCSantos, 2018-09-28:
Thanks @m4nh for your help. I will custom this parameter according to your suggestions.
Thanks a lot.

m4nh, 2018-09-28:
@BrunoEduardoCSantos  please post a screenshot when you resolve the problem (because i'm sure you will resolve it :) ) just for other user .. thanks a lot