m4nh/skimap_ros #28: Pose together with Points integration into the map

BrunoEduardoCSantos, 2018-10-25:
Hi,
I read the service code to adapt for my own purposes. Regarding the following piece of code,
```
tf::Transform base_to_camera = tf::Transform(
      tf::Quaternion(
          req.sensor_pose.orientation.x, req.sensor_pose.orientation.y,
          req.sensor_pose.orientation.z, req.sensor_pose.orientation.w),
      tf::Vector3(req.sensor_pose.position.x, req.sensor_pose.position.y,
                  req.sensor_pose.position.z));

  tf::Quaternion q = base_to_camera.getRotation();
tf::Vector3 v = base_to_camera.getOrigin();
```
Does this transformation helps improving the mapping ? If so, could you detail how does it work in general terms?
Thanks a lot.
Regards,
Bruno

m4nh, 2018-10-25:
@BrunoEduardoCSantos  Without this transformation the procedure can not really work if each cloud point is in the camera reference system, because with this transformation the points are transformed into the world reference system. If instead each cloud point is already in the world reference system, this transformation is useless and should not be done.

BrunoEduardoCSantos, 2018-10-25:
Thanks a lot @m4nh  :+1:

closed 2018-10-25 by BrunoEduardoCSantos