m4nh/skimap_ros #31: SIGDEV running skimap_live

BrunoEduardoCSantos, 2018-10-29:
Hi @m4nh ,
The node dies after a sometime.
Debugging using GDB I got the following error:

```
> Thread 1 "skimap_path_pla" received signal SIGSEGV, Segmentation fault.
> 0x00000000004755db in skimap::SkipListMapV2<skimap::VoxelDataRGBW<short, unsigned short>, short, float, 8, 8, 8>::integrateVoxel(short, short, short, skimap::VoxelDataRGBW<short, unsigned short>*) ()
```
I guess is something related with threading and OpenMP, but I am not sure.
Thanks in advance,
Bruno

m4nh, 2018-10-29:
@BrunoEduardoCSantos  it is useful to obtain a full debug stack to understand the problem.

BrunoEduardoCSantos, 2018-10-29:
```
[New Thread 0x7fffed930700 (LWP 13076)]
[New Thread 0x7fffed12f700 (LWP 13077)]
[New Thread 0x7fffec92e700 (LWP 13078)]
[New Thread 0x7fffe7fff700 (LWP 13090)]
[ INFO] [1540826627.666582571]: main: instantiating an object of type ExampleRosClass
[ INFO] [1540826627.667948509]: Initializing Subscribers
[ INFO] [1540826627.673202219]: main: going into spin; let the callbacks do all the work
[New Thread 0x7fffe77fe700 (LWP 13157)]
[New Thread 0x7fffe6ffd700 (LWP 13647)]
[New Thread 0x7fffe67fc700 (LWP 13648)]
[New Thread 0x7fffe5ffb700 (LWP 13649)]

Thread 1 "skimap_path_pla" received signal SIGSEGV, Segmentation fault.
0x0000000000475561 in skimap::SkipListMapV2<skimap::VoxelDataRGBW<short, unsigned short>, short, float, 8, 8, 8>::integrateVoxel(short, short, short, skimap::VoxelDataRGBW<short, unsigned short>*) ()
```

BrunoEduardoCSantos, 2018-10-29:
@m4nh  like this? Or is there any gdb option to provide more detail?
Thanks.

BrunoEduardoCSantos, 2018-10-30:
```
---Type <return> to continue, or q <return> to quit---
ng, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&>&, boost::_bi::list1<boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&>&, int) ()
#5  0x000000000048392f in void boost::_bi::bind_t<void, boost::_mfi::mf1<void, SkimapPathPlanning, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&>, boost::_bi::list2<boos
t::_bi::value<SkimapPathPlanning*>, boost::arg<1> > >::operator()<boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&>(boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocat
or<void> > const> const&) ()
#6  0x0000000000480594 in boost::detail::function::void_function_obj_invoker1<boost::_bi::bind_t<void, boost::_mfi::mf1<void, SkimapPathPlanning, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator
<void> > const> const&>, boost::_bi::list2<boost::_bi::value<SkimapPathPlanning*>, boost::arg<1> > >, void, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&>::invoke(boost:
:detail::function::function_buffer&, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&) ()
---Type <return> to continue, or q <return> to quit---
#7  0x000000000048848d in boost::function1<void, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&>::operator()(boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<vo
id> > const> const&) const ()
#8  0x00000000004860b6 in boost::detail::function::void_function_obj_invoker1<boost::functio
#7  0x000000000048848d in boost::function1<void, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&>::operator()(boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<vo
id> > const> const&) const ()
#8  0x00000000004860b6 in boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&)>, void, boost::shared_
ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const>) ()
#9  0x000000000048f2d0 in boost::function1<void, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> >::operator()(boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> >
const>) const ()
#10 0x000000000048ca6d in ros::SubscriptionCallbackHelperT<boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&, void>::call(ros::SubscriptionCallbackHelperCallParams&) ()
#11 0x00007ffff78c5d8d in ros::SubscriptionQueue::call() () from /opt/ros/kinetic/lib/libroscpp.so
#12 0x00007ffff786b838 in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) () from /opt/ros/kinetic/lib/libroscpp.so
#13 0x00007ffff786d23b in ros::CallbackQueue::callAvailable(ros::WallDuration) () from /opt/ros/kinetic/lib/libroscpp.so
#14 0x00007ffff78c9e39 in ros::SingleThreadedSpinner::spin(ros::CallbackQueue*) () from /opt/ros/kinetic/lib/libroscpp.so
#15 0x00007ffff78aee9b in ros::spin() () from /opt/ros/kinetic/lib/libroscpp.so
#16 0x00000000004692fd in main ()
```

m4nh, 2018-10-30:
@BrunoEduardoCSantos this is not related with Skimap, it is a bug related to your code.

BrunoEduardoCSantos, 2018-10-30:
@m4nh  Related with PointCloud2Message problem?

BrunoEduardoCSantos, 2018-10-30:
Thanks for your feedback.

BrunoEduardoCSantos, 2018-10-30:
So, I found the solution: the map resolution was too low. The resolution should be at least 0.05.

closed 2018-10-30 by BrunoEduardoCSantos