m4nh/skimap_ros #35: savetofile for

shadowver, 2019-01-29:
@m4nh Thanks for the great work! As I gone through the code, the SaveToFile() seen like only store the points. I  curious what is output file format and is the output file only suitable for rviz other than other rendering engine? Thanks.

m4nh, 2019-01-29:
hi @shadowver ! The SaveToFile() function is only a helper function useful to store a PointCloud-like representation of the voxel grid. What rendering engine do you mean? Something similar to [MagicaVoxel](https://ephtracy.github.io/)?

m4nh, 2019-01-29:
@shadowver actually you can produce a mesh from the voxel grid creating a cube for each point. But it is not optimized. You can follow this [Tutorial](http://ilkinulas.github.io/development/unity/2016/04/30/cube-mesh-in-unity3d.html
) about cube mesh generation starting from a seed point:

shadowver, 2019-01-30:
Thanks a lot! I guess it is the so far the most feasible solution. Closing this issue now =)

closed 2019-01-30 by shadowver